diablodale/dp.kinect #23: The Kinect attached is an unsupported model.

matkinect, 2015-02-04:
Hi Diablodale,

This is my config:
-Windows 7 64 biton ASUSTek COmputer
-Intel(R) Core(TM) i7-3610QM CPU @ 2.30GHz
-12G Ram
-KinectRuntime-v1.7
-Max7_x64_141202_57afb55

I'm using a refurbished Kinect  (that's what the box says) but it's new... It says model 1414. I can give you the serial if you need.
All drivers seem to be correctly installed on the device manager ( I can see 4 of them under Kinect subgroup: 
Kinect for Windows Audio array control
Kinect for Windows Camera
Kinect for Windows Device
Kinect for Windows  Security control)

I used the 64 bit version of your dp.kinect.mxe64 and the dp.kinect.mxe version too.
I start MAX by click on either dp.kinect.mxe64 or dp.kinect.mxe, Max opens. I load the 
dpkinect.maxpat. 
I plug my kinect and I "click to start", this message appears (this log is with the dp.kinect.mxe):

dumpout2: lostkinect USB\VID_0409&PID_005A\5&371EDC02&0&3 USB\VID_045E&PID_02AE\A00362A14523122A
dumpout1: lostkinect USB\VID_0409&PID_005A\5&371EDC02&0&3 USB\VID_045E&PID_02AE\A00362A14523122A
The Kinect attached is an unsupported model.

If I plug the Kinect first and then load (same method) MAX and click to start:

I have :
dp.kinect: No Kinect chosen for output.

I click on the Skeleton and skeletonformat box on the right, I have :
dp.kinect: Kinect has not been successfully opened and initialized.
until I unclick.

What should I do doc?

I know that my Kinect device works because I've tried with other drivers on linux.
Thanks for your help !

Cheers,
Mat

Here's my detailed config:
{
	"version" : "Version 7.0.1 (57afb55) (64-bit windows)",
	"platform" : "windows",
	"arch" : "x64",
	"osversion" : "Microsoft Windows 7 Home Premium Edition Service Pack 1 (build 7601), 64-bit",
	"samplerate" : 44100,
	"iovs" : 512,
	"sigvs" : 64,
	"scheduler_in_audio_interrupt" : "off",
	"audio_drivername" : "ad_mme",
	"audio_driver_subname" : "",
	"license" : "demo",
	"eventinterval" : 2,
	"overdrive" : "off",
	"mixerparallel" : "off",
	"mixercrossfade" : 0,
	"mixerlatency" : 30.0,
	"mixerramptime" : 10.0,
	"videoengine" : "qt",
	"packages" : 	{
		"Beap" : 		{
			"name" : "BEAP",
			"type" : "package",
			"author" : "Matthew Davidson, Peter McCulloch",
			"description" : "BEAP (Berklee Electro Acoustic Pedagogy) Modular is a synthesis pedagogical tool.",
			"version" : "Master from GitHub",
			"major_version" : "0",
			"minor_version" : "0",
			"min_max_version" : "612",
			"min_osx_version" : "None",
			"min_win_version" : "None",
			"website" : "https://github.com/stretta/BEAP/wiki",
			"link_mac32" : "https://github.com/stretta/BEAP/archive/master.zip",
			"link_mac64" : "https://github.com/stretta/BEAP/archive/master.zip",
			"link_win32" : "https://github.com/stretta/BEAP/archive/master.zip",
			"link_win64" : "https://github.com/stretta/BEAP/archive/master.zip",
			"relative_path" : "None",
			"path" : "C:/Program Files/Cycling74/Max7/resources/packages/Beap"
		}
,
		"Gen" : 		{
			"name" : "Gen",
			"path" : "C:/Program Files/Cycling74/Max7/resources/packages/Gen"
		}
,
		"Max for Live" : 		{
			"name" : "Max for Live",
			"path" : "C:/Program Files/Cycling74/Max7/resources/packages/Max for Live"
		}
,
		"MaxDefaultContent" : 		{
			"name" : "MaxDefaultContent",
			"path" : "C:/Program Files/Cycling74/Max7/resources/packages/MaxDefaultContent"
		}
,
		"MaxIntroLessons" : 		{
			"name" : "MaxIntroLessons",
			"path" : "C:/Program Files/Cycling74/Max7/resources/packages/MaxIntroLessons"
		}
,
		"vizzie" : 		{
			"name" : "vizzie",
			"path" : "C:/Program Files/Cycling74/Max7/resources/packages/vizzie"
		}
,
		"Max" : 		{
			"name" : "Max"
		}
,
		"MSP" : 		{
			"name" : "MSP"
		}
,
		"Jitter" : 		{
			"name" : "Jitter"
		}

	}

}

diablodale, 2015-02-04:
Microsoft does not license or support the Kinect for XBox (model 1414) in the manner in which you are using it. My wiki https://github.com/diablodale/dp.kinect/wiki has the requirements/setup and links to Microsoft's licensing discussion on this topic at https://msdn.microsoft.com/en-us/library/hh855358.aspx

You best and most flexible solution for this is to use the new Kinect for Windows version 2.

closed 2015-02-04 by diablodale

diablodale, 2019-08-03:
Microsoft lost SDK docs for v1 sensor :-(
Here is an archive link that is working for now http://web.archive.org/web/20140608214838/https://msdn.microsoft.com/en-us/library/hh855358.aspx