diablodale/dp.kinect #24: Nor trial or licence reg key works

augustely, 2015-03-18:
Hi, I've just purchased a licence for dp.kinect but I can't register.
I've done everything you said and max says:
dp.kinect: dp.kinect is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.

i've also tried the trial reg key but I have the same messages.

Can you help me with that please?
thanks

diablodale, 2015-03-19:
To help you, I would need a full issue report. Please read pinned Issue #18 to see how to provide a full report.
Until then, please be sure you are downloaded the registration key from the email. AND using the private registration string also from the email.

diablodale, 2015-03-28:
closing due to no reply.

closed 2015-03-28 by diablodale