diablodale/dp.kinect #26: Not staying registered

lwoodbury, 2015-05-06:
Windows 7 Pro 64b
Kinect SDK v1.8
Max 6.1.10 Runtime (x86)

1. I put the purchased registration file in a folder on a networked drive
2. I chose this file from the help patch as per the instructions here: https://github.com/diablodale/dp.kinect/wiki 
3. the Max window said the object was now registered, but on restarting the patch it says it is unregistered or expired.

diablodale, 2015-05-06:
A few more details please.
1. What is the "b" in 64b?
2. Is the computer's time correct?
3. What are the steps that you take to run Max, open the patch, do you doble-click the patch or use file/open, etc.
4. What steps do you take after you register and "restarting the patch".  What is the patch...is it the demo patch or a patch of your own? How do you open that patch? double-click, file/open, etc.
5. Is there any other version of Max installed on this computer? Or is Max 6.1.10 Runtime x86 the only version installed?
6. Please enable "@verbose 1" on the parameter line of the dp.kinect object that you use for the registration. Save that patch so that the verbose=1 is saved with it.
Then run Max 6.1.10 runtime x86.
file/open the registration patch
Do you have errors in the Max console about registration? If yes, copy all lines from the Max Console into a file.
Now register dp.kinect
Copy all lines from the Max Console into a file.
close the patch
close Max
Then run Max 6.1.10 runtime x86.
file/open the registration patch
Do you have errors in the Max console about registration? If yes, copy all lines from the Max Console into a file.
7. Paste in here all the lines from the Max Console that you captured or put a link here to that content.

lwoodbury, 2015-05-06:
The dp.kinect version was 1.0.2.0, I updated to the latest version (1.0.6) and re-registered and it all seems good now. Thanks for your help.

closed 2015-05-06 by lwoodbury

diablodale, 2015-05-06:
Good to hear you resolved the problem. :-)
This might have been related to some registration code improvements in version 1.0.5.