diablodale/dp.kinect #27: dp.kinect v1.0.6.0 standalone

silicatproject, 2015-10-01:
Hello,

Win7 
i7 4710 2.5ghz
16go ram
gtx 980m
max 7.05

I'm trying to build a standalone with dp.kinect v1.06.0 without success .

MaxConsole print : 

" dp.kinect v1.0.6.0, Copyright (c) 2011-2014 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those: a) documented in the dp.kinect terms at http://hidale.com,
-and b) granted by purchase or agreement to any entity listed as registered below.
-Includes some derivative work based on (c) Microsoft source code licensed under Apache License Version 2.0.
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect registered to "XXX XXX" Valid 1 Jul 2014 - 31 Dec 2999.
dp.kinect requires dp.kinectsupport_32.dll 1.7.0.510 or greater
Gen working in runtime mode "

Here a CopyScreen of the program

[image]

Hope you can help me .
Mathieu

diablodale, 2015-10-01:
I just tried and that works with Max 6.1.9 x86
I made a patch with only a bang and dp.kinect.
saved it
file/build coll+app
clicked include file
choose dp.kinectsupport_32.dll 
click build
save as=application
clicked save

Then I was able to run the EXE it built and the patch was ok.

silicatproject, 2015-10-05:
Great 
thanks so much ,
It works

closed 2015-10-05 by silicatproject