diablodale/dp.kinect #29: Registration key fail windows 8 64 bit

fredalemany, 2015-10-20:
This is my configuration :

1 Windows 8 64 bit

2 MAX MSP 6.1 64 bit registered

3 dp.kinect v1.0.6.0, Copyright (c) 2011-2014 Dale Phurrough 

4 registration name : dp.kinect-527

5 Max console message :
Jitter initialized
dp.kinect v1.0.6.0, Copyright (c) 2011-2014 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those: a) documented in the dp.kinect terms at http://hidale.com,
-and b) granted by purchase or agreement to any entity listed as registered below.
-Includes some derivative work based on (c) Microsoft source code licensed under Apache License Version 2.0.
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.
dp.kinect: dp.kinect is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.
dumpout1: open 0

6 time correct

Thanks

diablodale, 2015-10-20:
Please refer to the registration section of the wiki:  https://github.com/diablodale/dp.kinect/wiki#register-dpkinect

You will need the registration key file you downloaded + the registration name you were sent in email.
dp.kinect-527 is not your registration name.

Please try again using the linked process and the registration name you were sent in email.

fredalemany, 2015-10-20:
Thank you, il will try with the good name found in the email !

Thanks ...

Fred

fredalemany, 2015-10-21:
Ok all is good with the regsitration name send in the email ...

Thanks a lot for your quickly help ...

diablodale, 2015-10-21:
Happy to hear all is ok.
Thank you for choosing dp.kinect for your solutions.

closed 2015-10-21 by diablodale

fredalemany, 2016-04-07:
Mesdames,

Messieurs,

Le 27 Delvalle NICE vous invite à la rencontre de présentation de la
formation Simplon, *le lundi 18 avril 2016 à 11h*, en présence de Mme
Dominique Estrosi-Sassone, Adjointe au Maire, Déléguée au Logement, à la
Rénovation urbaine et à la Proximité, et du Pr Olivier Guérin, Délégué à la
Santé, à la prévention et à l’accompagnement de la perte d’autonomie.

L'occasion de rencontrer tous les partenaires de ce projet, la ville de
Nice, la métropole NCA, l'école de coding *Simplon <http://www.simplon.co/>*
et l’association le Hublot :

- pour évaluer vos besoins de formations,
- pour anticiper  vos prochains recrutements,
- pour dynamiser vos projets d'innovations services et web,
- pour soutenir le financement de cette première promo à Nice,

Merci de bien vouloir confirmer votre présence en cliquant sur le lien
suivant :
https://www.eventbrite.fr/e/billets-simplon-au-27delvalle-24340164086

Sincères salutations.

*Le 27 Delvalle** Santé connectée, 27* rue du Professeur *Delvalle*
06000
*Nice Pasteur ...*

[image: simplon.jpg]

<http://lsi.nicecotedazur.org/r/1/425d7f59-8ccb-4241-afd1-768fc9459711/undefined>

*Franck VIANO*
Responsable Communication et Evénementiel
Direction des Relations Publiques

METROPOLE NICE COTE D'AZUR  - 06364 Nice cedex 4
Tél : 0497133585 / Mobile : 06 61 36 68 01 / Fax : 4725
franck.viano@nicecotedazur.org

<http://lsi.nicecotedazur.org/r/8/a2f84c1e-b7ec-44f6-8df9-505b7dbf5ecb/undefined>

​Le Hublot Nice

http://www.lehublot.net

info@lehublot.net

fredalemany, 2016-07-13:
Tout Azimuth

*Espace de rencontre assistée par ordinateur*

*De l’installation comportementale au bal interactif*
La Guinguette
<http://www.culture-13.fr/agenda/la-guinguette-spectacles-gratuits-en-plein-air-35494.html>-
Spectacles gratuits en plein air
Mardi 19 juillet à 22h Cité le Clos (13ème) MARSEILLE
organisé par l'Espace Culturel de la Busserine
04 91 58 09 27

Une installation d'art numérique qui révèle et amplifie les attractions
amoureuses et sociales dans l’espace public et vous invite à venir danser
sous un réverbère musical, le temps d’un bal interactif, poétique et
ludique.

​*Tout Azimuth <http://www.lehublot.net/azimuth/>Arts de la rue **/** Arts
numériques*
*, Création 2010*

*En vidéo  en 2008 au FAR de Morlaix ... <https://youtu.be/fbUfm6j80As>*

*Frédéric Alemany* conception programmation graphique et sonore

*Philippe Maurin* scénographie dramaturgie

*Benoît Seyrat* musique

*Nathalie Cannatella*  danse

*Production** Le Hublot*

Avec le soutien *du Conseil Artistique à la Création – Théâtre, du Conseil
Régional PACA*

*Coproductions 2009 : * *L'Entre-Pont (Nice) **Le Zinc (Marseille)** Le
Fourneau (Brest)*

*Contact diffusion : Association Diva – Le Hublot*

*Céline Roques*

*Tél. : 04 93 31 33 72 **/** 06 16 60 03 48 * *celine@lehublot.net
<celine@lehublot.net>*

Association Diva – Le Hublot  16 rue de Roquebillière 06300 Nice

fredalemany, 2016-09-29:
Du samedi 08 octobre 2016 au mardi 18 octobre 2016 ...
*Opposite* installation interactive création 2016
Le Hublot Nice, Frédéric Alemany

Opposite est une installation d'art numérique qui permet au public de jouer
avec les forces et les éléments essentiels à la formation d'un système
solaire.

​
Vidéo et photo Féeries nocturnes 2015
https://vimeo.com/141888836

Opposite cherche à rendre sensible le rôle des oppositions dans
l'organisation des formes physiques et esthétiques. C'est un processus de
recherche qui croise création numérique et culture scientifique sur les
forces d'attractions et de répulsions qui animent le monde physique des
atomes et des corps célestes.

En prenant support sur les vibrations réelles de l'eau pour dessiner des
motifs géométriques réguliers et en utilisant des algorithmes inspirés par
les simulations numériques en astrophysique pour générer le mouvement des
particules, l’installation d’Opposite rend visible et audible, plus
particulièrement, le rôle des résonances dans la formation et l’évolution
du système solaire.

Le public est invité à façonner avec ces mains cet univers imaginaire qui
peut à tout moment se détruire ou évoluer, se complexifier pour faire
apparaître turbulences et cycles réguliers. De la simplicité de ces
principes de simulation et par l'heureux hasard de la manipulation peuvent
ainsi naître une diversité de mondes possibles.

Cette oeuvre numérique poursuit la tradition de l’harmonie des sphères
croisant astronomie et esthétique pour renouveler ce cheminement artistique
au regard des avancées scientifiques actuelles.
https://sites.google.com/a/lehublot.net/opposite/
*Opposite **performance audio visuelle **# 3*

Lien vidéo performance # 3 https://youtu.be/nOtzyaJuXRU

​

-- 
Frédéric Alemany
06 16 60 03 48
fa@lehublot.net

fredalemany, 2017-03-27:
* SAMEDI 1er AVRIL  A L'ENTRE-PONT NICE*

* LE HUBLOT PROPOSE *

*LES RENDEZ-VOUS ARTS NUMERIQUES / MUSIQUES ELECTRO *
[image: Affiche_RDV_Arts_Numriques_1er_avril__500_pix]

*ADHESION PUBLIC - ANNEE 2017*
Adhésion individuelle annuelle de 2€ (carte valable jusqu'au 31/12/17)
+ participation financière à chacune des propositions
(BAL POP TRONIC : 5€ - DIMITRI FROM PANISSE - Gratuit ).
*Paiement par chèque ou espèce uniquement.*

*EN SAVOIR PLUS...*

*Buvette/Cantine sur place dès 20h *
Jacques Schaeller vous propose un balèti de polenta provençale et sa salade
de carottes marocaines à 10€ ou une pissaladière à 3€.

Réservation jusqu'au 27 mars
par SMS : 06 61 20 89 66 ou par mail : celine@lehublot.net

                                                 [image: Bal_pop_tronic]

*21h> BAL POP TRONIC*

*  balèti électronique  Delta Sònic, Jankenpopp et Tapetronic*

A la croisée des musiques traditionnelles et de l'électro retro-futuriste,
Bal Pop Trònic est la rencontre de trois musiciens partageant un intérêt
pour la lutherie sauvage et avant-gardiste: Game Boy détournée,
magnétophones modifiés et pipeaux de Camargue, etc. Ils vous entraînent
dans un bal inattendu de bourrée scratchée en farandole 8bit, pour le plus
grand plaisir des fans de trad et de dancefloor ! Avec Henri Maquet (alias
Delta Sònic, musicien et musicologue connu pour sa grande maîtrise du
galoubet et du smartphone), Pierre-Erick Lefebvre (alias Jankenpopp,
inventeur du son de demain et grand détourneur de circuits électroniques)
et Alexis Malbert (alias Tapetronic, père de la Scratchette - la cassette
qui scratche - et docteur ès gadgets sonores).

http://balpoptronic.tumblr.com/

                                                  [image:
Dimitri_from_panisse]

*22H> DIMITRI FROM PANISSE*

* DJ selectropical *

Dimitri a le don de rassembler des personnes d'horizons divers et de les
faire danser et se rencontrer. Après avoir été membre actif du collectif
Zou Maï, d'une batucada, de l'organisation du carnaval indépendant de Nice
ainsi que des fêtes traditionnelles
ré-inventées (Santa Capelina le 1er mai, Lo Paillassou, ...), il continue
la fête en organisant des soirées réunissant des groupes et des DJ's dont
lui avec des rythms groove & musica-ethnica.

>>Fermeture à minuit.

 *SUIVEZ L'ACTUALITE DE L'ENTRE-PONT...*

Sur le site *www.entrepont.net* <http://www.entrepont.net>

Sur la page facebook :

[image: facebook-logo-150x150] <https://www.facebook.com/lentrepont/>

L'Entre-Pont est soutenu par la Direction Régionale des Affaires
Culturelles PACA, le Conseil Régional PACA, le Département des
Alpes-Maritimes et la Ville de Nice.
L’Entre-Pont a signé une convention d'objectifs triennale avec la Ville de
Nice.

<http://entrepont.net/index.php/residences-accompagnees>

*L'Entre-Pont*
Accès : 89 route de Turin - 06300 Nice (aux anciens abattoirs)

Pour se rendre à l'Entre-Pont :

- Tram : arrêt "Vauban" puis 15 min à pied en remontant la route de Turin
jusqu'au n°89
 (grand portail jaune sur le trottoir de gauche).
- Bus 4 ou 6 : arrêt "Abattoirs".
- Station Vélos bleus n°154.

>>  Un parking est à disposition du public.

Tél. : 06 79 83 63 44 / Email : rp@entrepont.net <contact@entrepont.net>
Site : www.entrepont.net
Blog : http://memoirehallespada.over-blog.com

Se désabonner
<http://s359465721.onlinehome.fr/index.php?subid=3840&option=com_acymailing&ctrl=user&task=out&mailid=110&key=hkYmSeVPUCqOwn>

-- 
Frédéric Alemany
06 16 60 03 48
fa@lehublot.net

fredalemany, 2017-06-13:
Bonjour,

Le Hublot, centre de création numérique à Nice va bientôt finaliser sa
programmation de résidences et de rendez vous arts numériques.

Nous invitons les artistes à nous communiquer leurs demandes de résidences
simples ou accompagnées pour leurs projets de créations et aussi leurs
demandes de diffusions dans le cadre des nos rendez vous arts numériques.

Cette année le Hublot proposera un temps fort de programmation en novembre
2017 avec Movimenta, un festival sur l'image en mouvement en partenariat
avec l'Eclat.

Vous disposez pour vous manifester de deux formulaires :

Résidences
<https://docs.google.com/a/lehublot.net/forms/d/e/1FAIpQLSf_M0urYm17ealHKN2X0NtIAdr1Matfaj2a0eKDcAzqDjlREA/viewform?c=0&w=1>

<https://docs.google.com/a/lehublot.net/forms/d/e/1FAIpQLSc_xh32W2jlGxBoApf8l5cerwNzOdpKK4yEPjub2Go4fR5_rA/viewform?c=0&w=1>
Programmation
<https://docs.google.com/a/lehublot.net/forms/d/e/1FAIpQLSc_xh32W2jlGxBoApf8l5cerwNzOdpKK4yEPjub2Go4fR5_rA/viewform?c=0&w=1>

Merci bien.

-- 
Frédéric Alemany
06 16 60 03 48
fa@lehublot.net