diablodale/dp.kinect #31: Use french language with 'speech' dpkinect

loopiotte, 2016-02-20:
Hi, 

First, excuse for my bad English..

I'm a french user of dpkinect1 and i want to use french language with 'speech' .

I use MacBook pro retina with bootcamp and Windows 10.
Runtime1,8, 

i download the 'Kinect french language pack' but i don't understand how configure dpkinect for use this pack .

thanks for your response.

diablodale, 2016-02-20:
Please follow the dp.kinect2 setup guide
https://github.com/diablodale/dp.kinect2/wiki#setup-guide
and in particular step 8 for the speech support

Then on that same page read the wiki for the attributes and messages which support speech.
For example: https://github.com/diablodale/dp.kinect2/wiki/Attributes#audio-and-speech
https://github.com/diablodale/dp.kinect2/wiki/Message-based-Data#speech-recognition

If you have difficulty after reading the documentation above and referring to the working demo patch included in your ZIP download, then please provide all the information I request by referring to https://github.com/diablodale/dp.kinect/issues/18

diablodale, 2016-02-20:
Please refer to the dp.kinect setup guide
https://github.com/diablodale/dp.kinect/wiki#setup-guide
and in particular step 8 for the speech support

Then on that same page read the wiki for the attributes and messages which support speech such as:
https://github.com/diablodale/dp.kinect/wiki/Attributes#audio-and-speech
https://github.com/diablodale/dp.kinect/wiki/Message-based-Data#speech-recognition

If you have difficulty after reading the documentation above and referring to the working demo patch included in your ZIP download, then please provide all the information I request by referring to https://github.com/diablodale/dp.kinect/issues/18

diablodale, 2016-02-24:
Have you made progress or resolved your question?

loopiotte, 2016-02-26:
Hi,

I read the wiki but I dont understand how the dpkinect object use de the
differents librairy of language and if it's an obligation to write a
'grammarfiles'.

excuse me for my bad english..

thanks for your response

diablodale, 2016-02-26:
I can also answer in the German language if that is better for you. Ist Deutsch besser für Ihnen?

Yes. You must write a grxml grammar file. That file changes what the speech recognition hears -> messages you want your patch to receive. I recommend you use the links in my documentation to read about the grxml file. You can also google for grxml and probably find documentation in your native french language. (Or use google translate)

loopiotte, 2016-02-27:
Thanks you.

envoyé de mon sebsung.

closed 2016-04-07 by diablodale