diablodale/dp.kinect #32: no help file found for jbogus ???

CarloCattano, 2016-02-21:
i reopened dp.kinect after a while that i bought it and i get this message on the max console . Any idea whats wrong ??

diablodale, 2016-02-21:
I don't have enough information to comment on this. Please following the instructions at https://github.com/diablodale/dp.kinect/issues/18 to provide me the needed information to answer your inquiry.

CarloCattano, 2016-02-21:
-ACER LAPTOP I7 8GB RAM WINDOWS 10 64 BITS , HDD 1TB DRIVE . 
Installed Kinect 1.7 SDK
Installed Max 7.1 x64 and it is registered (MAX FOR LIVE LICENSE) (EDITING INSIDE LIVE)
ATTACHED SCREENSHOT 
[image]

i get randomly between this two messages: 
unable to find jbogus help file 
and dp.kinet requires dp.kinectsupport_64.dll 1.7.0.510 or greater.

The funny thing is that "sometimes" is opening and working , i really have no clue what can be the problem

Kinect *.mxe64 files are in the search path of max as well as in the externals directory in the max folder .

my max for live patch
https://www.dropbox.com/s/1lnuv6wvquq8jgg/kinect_dp_1.amxd?dl=0

diablodale, 2016-02-21:
Your MME and helpfile errors are not related to dp.kinect. There is something else in Max causing those problems.

The remaining third "dp.kinect requires xxxx" is telling you the exact issue. It needs that file.with that specific version or greater.
Please refer to the setup guide at https://github.com/diablodale/dp.kinect/wiki#setup-guide
Your issue is with steps 5-7. By resolve these steps, you can move past that error.

As helpful advice, I caution having multiple copies of externals in a Max search path. This is true for my externals and externals from others developers. It is very easy to create version conflicts. Please remove all extra copies and have only one dp.kinect in your search path.

If you want to have multiple, then you need to have steps 5-7 done the same for the given external and its specific version.

diablodale, 2016-02-24:
Have you resolved the missing dp.kinectsuppoert_64.dll error and dp.kinect working again?

CarloCattano, 2016-02-24:
Yes , i guess it was about the location of my patch.
I needed to remake the patch . i opened your dp_kinect patch then converted to a max4live midi effect in the same directory , deleted all the other copies of the externals and dll´s .
Now working but i have a new issue wich is that i cant visualize the jitter windows when i lock my patch and try to visualize it inside live in presentation mode . but i guess i should open a new thread or either not cause is more related to max than to you . thanks for your support.
p.d : i couldnt find in the directory the ranges that kinect is giving output  . As they are not body relative but relative to the distance to the kinect . Im trying to find the ranges to scale the data to midi values , as im very amateur in max i wonder if maybe someone already did something similar.

diablodale, 2016-02-24:
good to hear you resolved your issue.
All jitter use/visualization or Ableton Live use is supported by Cycling74 and/or Ableton.
Try the Cycling74 forum for Max for Live. They are very good resources. Here is a link:
https://cycling74.com/forums/forum/max/#gsc.tab=0

All the data values for dp.kinect output are in the wiki documentation starting at https://github.com/diablodale/dp.kinect/wiki
Here is a quick and simple example:
* The Kinect v1 sensor has a body tracking range of 0 to 4 meters in perfect condtions
* The data output by the "skel" message is in meters or millimeters (it is your choice) https://github.com/diablodale/dp.kinect/wiki/Message-based-Data#skeleton-joints
* You can capture this skel message. "route" or "routepass" are good for this. See the demo patch I provide for examples.
* Then use standard Max objects like "scale" or "zmap" to do a simple mapping from 0-4 meters -> 0-127 midi
Max has built in documentation and tutorials so you can see examples of scaling data and using data with MIDI.

Happy patching! And thank you for choosing dp.kinect for your solutions.

closed 2016-02-24 by diablodale

CarloCattano, 2016-02-24:
thanks so much for the information. im still very green on max but learning a lot lately. thanks for your time