diablodale/dp.kinect #33: 'data unavailable for request

johnnyvenom, 2016-06-22:


johnnyvenom, 2016-06-22:
Update: as a troubleshooting measure, I uninstalled Runtime v1.8 and installed Runtime v1.7. With 1.7, the Max console gives me the following error when I plug in the Kinect: 

    dumpout2: lostkinect USB\VID_045E&PID_02C2\5&330350CB&0&5 USB\VID_045E&PID_02AE\0000000000000000
    dumpout1: lostkinect USB\VID_045E&PID_02C2\5&330350CB&0&5 USB\VID_045E&PID_02AE\0000000000000000
    The Kinect attached is an unsupported model.

Also, with v1.8, the skeleton tracking is working with the example apps in the Kinect Developers SDK bundle. 

Hope that helps.

johnnyvenom, 2016-06-22:
OK, looking at other documentation and closed issues, this is because of the Kinects I am using - They are XBox 360 Kinects, model 1414 and 1473. So if I understand correctly, even though they give the IR and RGB camera outputs, they are not supported for the skeletal tracking?

diablodale, 2016-06-23:
Hello. Thank you for considering dp.kinect for your solutions.
My setup page at https://github.com/diablodale/dp.kinect/wiki#terms-of-use-licensing-conditions-and-warranty addresses this and the link I provide to Microsoft. When you get "The Kinect attached is an unsupported model", that is an error Microsoft reports and enforces related to their licensing.
In my dev testing, I have used one of the 360 models. I don't have accessible to check a model #.
An alternative not addressed above are Kinect for Windows (v1 and v2) sensors. They are quality, more features, and more stable than the 360's, and available for cheap.
If Microsoft supports your sensor on your hardware, then I also support it. Last I read, Microsoft doesn't support the Kinect for Windows v1 on Windows 10; though I heard rumor from a few people that they got it working on Windows 10 w/ the 1.8 runtime.
Your hardware scenario is outside anything I heard Microsoft officially supports (mac, win10, Kinect 360). Therefore, you are likely in untested territory.
If the above doesn't address your issue, please follow https://github.com/diablodale/dp.kinect/issues/18

johnnyvenom, 2016-06-23:
Thanks for the information and prompt response. That all makes good sense, so I'll consider the issue closed. If I make progress or find a path to success I'll let you know. Cheers!

closed 2016-06-23 by johnnyvenom

diablodale, 2016-06-23:
Yes, please do. I aim to provide quality software.
At the Microsoft link, they do write about some licensing exceptions for developers. This might be a possibility for you. Though the mac, win10 compatibility continues.
FYI, I use Runtime 1.7 on Win10 x64 with no issues. I've also used runtime 1.8