diablodale/dp.kinect #35: Problem with multiple kinects

freqshift, 2016-10-01:
Toshiba P855 Laptop with 8GB RAM
Windows 7
Intel(R) Core(TM) i7-3630QM 
Nvidia GForce GT 640 M

Kinect runtime 1.7
Microsoft Kinect SDK 1.8
Max/MSP 6.1 and Max/MSP 7.1 (Both registered)

Both of the kinect xbox 360 1414 sensors work fine with the example patch if I use them separately, but I can not get them to work simultaneously. 
When I try to open the USB port of the second kinect device, all the streams seem to freeze. If I close one of the ports the other one starts to function again. The problem appears regardless to the version of Max.

diablodale, 2016-10-02:
What version of dp.kinect are you using?

Would you please provide a patch and steps to follow so that I can reproduce your issue? Or if you are using the demo patch included with v1.0.6, just provide me what steps to open patches, click buttons, etc.

Have you tried this with retail Kinect for Windows v1 sensors? The Microsoft drivers do operate differently on model 1414 with some things (e.g. camera control is not available and some combinations of depth/color resolutions are not available). I'm not confident that I did testing on two simultaneous 1414 sensors because of the Microsoft feature and licensing restrictions in using them. I'll look to see if I own two and can try to reproduce your issues. I'll also with the KfW v1's.

freqshift, 2016-10-02:
Im using v1.0.6
I only have the xbox360 1414 sensors, I haven't tried with KfW. Im using the demo patch with the folllowing steps:
1.Open the patch with the two kinects connected to the machine.
2.Double click on the subpatch called "more features"
3.Click on getusbidlist message
4.Start/open one of the dp.kinect instances (That works fine)
5.Start/open the other instance (This is when the problem occurs.)

Update: The problem occurs when I try to get color image AND skeleton/tracking data from both of the kinects simultaneously. Only color image works fine, and I guess only skeleton/tracking data works fine as well, furthermore I think only color image on one device and color+skeleton on the other works fine too. (I am still fiddling around.)

diablodale, 2016-10-02:
Suspicious. Could be a bug in code. I can look for it this week.

One more thing to try... please make sure that the two Kinect v1 sensors are not pointed in the same direction. For example, point them 180 degrees apart so there is no possibility for their lasers and image sensors to overlap. Then see if you have any problems.

Why? The Kinect v1 depth/skeleton sensors can often fail if two sensors overlap the same area. It has to do with the timing of the lasers and the invisible dots they draw. Here is an article on the topic https://www.precisionmicrodrives.com/tech-blog/2012/08/28/using-vibration-motors-microsoft-kinect

freqshift, 2016-10-02:
I have tried it, but they produce the same error. The article is great though, it can be very useful in the future, thanks! Meanwhile I have stumbled on another problem. I'm going to make a new issue for it.

diablodale, 2016-10-04:
I am not able to reproduce any problem. I have two XBox 360 model 1414's working well simultaneously.
Mine is a Windows 10 x64
Runtime 1.7
Max 6.1.9 x64  -and-  Max 7.2.4 x86

Steps:
1. used the demo patch of v1.0.6
2. opened the subpatcher
3. I disabled speech and facetracking because I don't have those installed.
4. I clicked getusbislist
5. I clicked the toggle in the subpatcher to start the kinect/patch
6. I clicked the toggle in the top patch to start the kinecct/patch

Result
All streams in top patch are working: I see depth, I see color, I see playermap, I see my xyz coordinates changing.
All streams in sub patch are working: I see color, I see point cloud, I see the red ball follow my right hand, I see the sound direction matching my location.

Please turn on @verbose and try again. Be sure to watch for any red errors in the Max console window. If you get suspicious messages in the console window, please copy/paste them into this issue so I can see them.

Please ensure that you have your two Kinect on separate USB2 controllers. There is only enough bandwidth to have a single Kinect on each controller. Also notice on this thread that there are known issues with USB3 controllers (and their USB2 backwards compatibility) on Windows 7.
https://social.msdn.microsoft.com/Forums/en-US/b8ff7e7b-befe-426c-8eae-192a9aa3fa88/multiple-kinects-on-an-intel-usb-30-controller-are-failing?forum=kinectsdk

freqshift, 2016-10-05:
Interesting... On my machine I had to install the SDK, because it did not run only with the runtime as far as I can remember.
You were right. With @verbose on sometimes I get the following error message:
"Insufficient power for Kinect. Or, the Kinect hub and motor are connected but the camera is not"
I guess this sounds to be the problem.

diablodale, 2016-10-05:
That error condition is detected by the Kinect drivers and I provide it with the verbose diagnostics. It should never occur when the USB port and Kinect hardware are operating normally. I have sometimes seen this error when the Kinect is plugged into the USB port but it isn't plugged into the mains electricity. In all cases, this is an indication that the problem is outside dp.kinect.

Moving your two Kinects to different USB controllers is often possible on laptops. Many of them are made with multiple USB controllers. Often they are on different sides of the laptop. In Windows, you can use Device Manager to see. Run it and on the menu choose to View devices by connection. Then you have to search. On my laptop, the USB controllers are under the PCI Root. My laptop has three controllers (one internal for fingerprint sensor, one external USB2, one external USB3). Therefore, I am able to put one Kinect 1414 on the USB2 controller and the 2nd Kinect on the USB3 controller (has USB2 backwards compatibility).

freqshift, 2016-10-07:
Thanks a lot! I've checked it, and unfortunately my laptop has only 1 USB3 controller for all the ports. I think this causes the weird behavior. But as I mentioned, only depthmap does not demand so much bandwidth and its functioning perfectly.

closed 2016-10-07 by diablodale