diablodale/dp.kinect #36: Max/MSP with dp.kinect crashes when trying to implement targeted user tracking

freqshift, 2016-10-02:
Toshiba P855 Laptop with 8GB RAM
Windows 7
Intel(R) Core(TM) i7-3630QM
Nvidia GForce GT 640 M

Kinect runtime 1.8
Microsoft Kinect SDK 1.8
Max/MSP 6.1 and Max/MSP 7.1 (Both registered)
dp kinect v1.0.6

The goal is to track one user (only Center of Mass) and ignore the others. If the user is lost then choose the latest active user and keep on tracking. 
The patch performs more or less accurately, but after a while it suddenly crashes, both under max 6.1 32bit and max 7.1 64bit. 
The patch can be found here, you will have to change the @idkinect argument to fit it to your device port:
https://github.com/freqshift/Max-MSPkinect/blob/master/dpcrash.maxpat

diablodale, 2016-10-04:
I am not able to reproduce any crash or problem. The sensor and your patch are working. For five minutes I moved in and out of view of the sensor on both versions of Max. I saw the center of mass number changing. I also saw the userid changing.

I have two XBox 360 model 1414's plugged in. Of course, only one is used by your patch.
Mine is a Windows 10 x64
Runtime 1.7
Max 6.1.9 x64 -and- Max 7.2.4 x86

Steps:
1. used your patch. removed the usbid param.
2. increased the font size of the user pos message box and the userid spinner so I could easily see them.
3. turned on toggle and opened kinect.
4. Then walked in and out of view of the Kinect for 5 minutes.

Result
The user pos coordinates updated as I moved.
The userid displayed a single number when it saw me. When it didn't see me it rapidly displayed many numbers.

Please turn on @verbose and try again. Be sure to watch for any red errors in the Max console window before you crash. If you get suspicious messages in the console window, please copy/paste them into this issue so I can see them.

Please ensure that you have your two Kinects on separate USB2 controllers. There is only enough bandwidth to have a single Kinect on each controller. Also notice on this thread that there are known issues with USB3 controllers (and their USB2 backwards compatibility) on Windows 7.
https://social.msdn.microsoft.com/Forums/en-US/b8ff7e7b-befe-426c-8eae-192a9aa3fa88/multiple-kinects-on-an-intel-usb-30-controller-are-failing?forum=kinectsdk

You have a modern laptop. Did you enroll in the free Windows 10 upgrade? If you did, I highly recommend it as Windows 7 is no longer supported by Microsoft.

Your crash may not be in the dp.kinect code. It could be in Max, Kinect drivers, USB drivers, or something else. To know that, you will need to turn on some feature of Windows 7 that will provide you a crash report and in that report has the "call stack" of functions/code that led up to the crash. This kind of report is built into Windows 8+. I don't know how to do it in Windows 7 (maybe debugdiag ?) so you will need to do some research on your own. If you get a crash, that call stack will tell you who caused the crash; it is almost always the function at the "top". The meaning of top, of course, will depend on how the report is written. If you get this call stack report, then please provide it so I can see the sequence of events that led up to the crash and who caused it. You might also find some crash information in your Windows NT event logs http://www.cyberlink.com/support/faq-content.do?id=10449

freqshift, 2016-10-05:
I did not upgrade to windows 10, because most of my friends/collegues were not that much impressed by it at the time, and I had softwares installed which were not yet supported. (The updated version is probably much better).
Here I get the same error message with @verbose on:
"Insufficient power for Kinect. Or, the Kinect hub and motor are connected but the camera is not"
I've just read the thread and I am more and more convinced that you are right. The USB Host Controller causes the issue. Is this problem solved in Windows 10, or one must have 2 physical controllers built in?
Anyway, I had to take a different approach on the project since I need user tracking from top view (with kinects installed on the ceiling), so I am only using the only the depthmap, and doing the rest in max. Up till now everything works.

diablodale, 2016-10-05:
If you want two Kinect v1 sensors, you must have two separate USB2 controllers. The Kinect sensor uses the majority of the USB2 bandwidth. Therefore, there is not enough for two on the same USB2 controller. Windows 8+ has full support for USB3 (including its backwards compatible USB2 feature). It should be more reliable as Microsoft writes in their articles/posts. Personally, I have never had a problem with USB3 ports on Windows 8+.

That error condition (power and connectivity problems) is detected by the Kinect drivers and I provide it with the verbose diagnostics. It should never occur when the USB port and Kinect hardware are operating normally. I have sometimes seen this error when the Kinect is plugged into the USB port but it isn't plugged into the mains electricity. It can also occur when the Kinect doesn't have full operation on the USB port. In all cases, this is an indication that the problem is outside dp.kinect.

Moving your two Kinects to different USB controllers is often possible on laptops. Many of them are made with multiple USB controllers. Often they are on different sides of the laptop. In Windows, you can use Device Manager to see. Run it and on the menu choose to View devices by connection. Then you have to search. On my laptop, the USB controllers are under the PCI Root. My laptop has three controllers (one internal for fingerprint sensor, one external USB2, one external USB3). Therefore, I am able to put one Kinect 1414 on the USB2 controller and the 2nd Kinect on the USB3 controller (has USB2 backwards compatibility).

One time, I experimented with putting Kinects on the ceiling and pointing them down. Then using filtering of the depthmap with a jit.op to isolate people from the floor. I also experimented using cv.jit.blobs.**** for this.

Happy patching!

freqshift, 2016-10-07:
Thanks a lot! I've checked it, and unfortunately my laptop has only 1 USB3 controller for all the ports. I think this causes the weird behavior. But as I mentioned, only depthmap does not demand so much bandwidth and its functioning perfectly.
For the tracking Im using the brilliant patch that involves cv.jit.label, blobs, and a java script object by Zachary Seldess:

https://cycling74.com/forums/topic/sharing-kinect-depth-sensitive-blob-tracker-3d-bg-subtraction-etc/

I had to modify it a little since I need quite a steady and reliable labeling for my interaction model, but basically this is the heart of the tracking engine.

closed 2016-10-07 by diablodale