diablodale/dp.kinect #37: "No kinect chosen for output"

karlskene, 2016-11-30:
My Setup
ASUS ROG laptop with 24 GB RAM, 256GB SSD, 1TB hard drive, Intel i7-4720HQ CPU @ 2.60GHz
Windows 10 64 bit
Windows 10 x64 with all current updates

-Installed Kinect 1.8 SDK
-Installed Max 7 (64 Bit) and it is not registered (part of Max for Live)
-Using dp.kinect 1.0.6.0
-Kinect v1 plugged into usb 2.0 controller
-downloaded and extracted dp.kinect and placed all files in the Max 7 Packages folder
-Opened the dp.kinect patcher in Max 7 and registered successfully
-when clicking the x box labelled "click to start" I get this message in the Max Console:
dp.kinect - no kinect chosen for output
dumpout1 0 open 0

diablodale, 2016-11-30:
Please read all the closed issues on this error. This is always a hardware/driver problem
https://github.com/diablodale/dp.kinect/issues?utf8=%E2%9C%93&q=is%3Aissue%20is%3Aclosed%20chosen

If you continue to have problems after reviewing all those issues, please provide the model number of your Kinect. This is found printed near the serial number. Also please turn on @verbose and provide all the output of the entire Max console.

karlskene, 2016-12-05:
> 93&q=is%3Aissue%20is%3Aclosed%20chosen

Hi Dale, I'm still having this issue after reviewing all those issues.  My
kinect model number is 1414.  When I turn on Verbose I get a repeating
message saying, "dp.kinect - Kinect has not been successfully opened and
initialized.

diablodale, 2016-12-05:
Microsoft's licensing terms limits Kinect model 1414 as I wrote in several of those closed issues. I suspect the issue you are experiencing is that you are trying to use that Kinect outside their licensing terms. I have no control of this as this is a Microsoft/driver issue as it relates to your license with them.
https://github.com/diablodale/dp.kinect/wiki#terms-of-use-licensing-conditions-and-warranty

diablodale, 2016-12-05:
Do the Microsoft samples that come with the SDK work? You can use the Kinect for Windows Developer Toolkit v1.8 that was installed with your SDK to run precompiled samples. Try "Kinect Explorer-D2D"
https://msdn.microsoft.com/en-us/library/hh855373.aspx

karlskene, 2016-12-06:
When I try to open the solution in visual studio it says, "Kinect
Explorer-D2D (unavailable)"

diablodale, 2016-12-06:
There is something failing with your PC, Windows, Kinect drivers, or the SDK. Support for all of them is provided by Microsoft. I won't be able to help you with those issues.

karlskene, 2016-12-06:
Strange, because it works with other software, like Z Vector.

diablodale, 2016-12-06:
Z Vector uses hacked (freenect) or illegal (openni 1.x) drivers. I know the Kinect very intimately and when on their support site they write it doesn't work well with Kinect model 1473...I know the underlying technology they are using.

dp.kinect is written using the official Kinect SDK from Microsoft. I support dp.kinect. Microsoft supports the Kinect, its drivers, and their Windows OS.

I am almost sure that your issue is still hardware/driver. It is that you are trying to use an XBox 360 Kinect against the licensing that Microsoft provides you. I provided you links above which specifically write about this. This is between you and Microsoft; something that I can't do anything about. If you are a developer, then there are methods (same link) that you can use an XBox 360 Kinect for testing and development....not for production usage.

When you are in compliance with Microsoft's licensing terms, Microsoft drivers will allow usage of the Kinect. I am not in control of that. Only Microsoft. Once Microsoft allows their drivers to work, dp.kinect will be able to find the Kinect and all will be well.

diablodale, 2016-12-08:
Checking back with you. Did you resolve the hardware or SDK issue?
If you did, is dp.kinect working for you now?

diablodale, 2016-12-13:
Closing this issue as it is by Microsoft's design.
If you have future questions, please open a new issue so that I can assist.

closed 2016-12-13 by diablodale