diablodale/dp.kinect #38: Hi Dale I have a problem with unregister and register

kenbyers, 2016-12-05:
Win 7,m Max 7 3.1

I want to register my dp.kinect on new computer, so that I can use it. The problem is that I cant unregister on old, and register on new computer does not open.
Could you offer a solution please?

Thanks 

Ken Byers

diablodale, 2016-12-05:
Hello. I have no information on which to answer your inquiry.
Please read https://github.com/diablodale/dp.kinect/issues/18 and provide all the information it guides you to do. After that, I will be able to understand your question and start on resolving it for you.

kenbyers, 2016-12-05:
I have rebooted a computer where dp2.kinect was on win 7, with SDK 1.8 and forgot to unregister.
Now on a different computer running Win 10, with the SDK 2.0, the registration button does not open for dp2.kinect. I have SDK 2.0 installed.

Thanks 

Ken

diablodale, 2016-12-05:
Unfortuntely, little of that makes sense to me. AGain, please read the link I provided above and provide all the information requested there. I will not be able to assist you until you provide me the information needed.

diablodale, 2016-12-07:
Is your issue resolved? If not, please read the link I provided above and provide all the information requested.

kenbyers, 2016-12-12:
Hi Dale  
My issue has been resolved.

Thanks

closed 2016-12-12 by kenbyers