diablodale/dp.kinect #39: registration not taking

jjolivette, 2017-01-03:
howdy,
i just purchased a license, placed the key in the dp.kinect2 help folder
and attempted to register with the name sent via email.
Max7 (on Win 10 pro) throwing an error saying that the object isn't registered 
or that the license has expired.
please advise

cheers!

diablodale, 2017-01-03:
Hello. thank you for choosing dp.kinect for your solution.
I am sure this is something easy to resolve. Please review the closed issues in this same issue database for similar problems. You will likely find your answer there or by closely following the setup guide again. Here is a link to the specific section on registration https://github.com/diablodale/dp.kinect/wiki#register-dpkinect

If you continue to have problems please review these instructions https://github.com/diablodale/dp.kinect/issues/18 and provide all the needed information here.

Thank you again

jjolivette, 2017-01-03:
thank you, sir

i’ve gone through the instructions again and don’t think i’m missing anything.
i’ve updated the ticket and am including the details of the request below. 
thank you for your help

Issue: 

the dp.kinect2 object (v1.12) refuses to register.
the license file is placed in the object’s help folder.
having entered the registration name, and clicked enter
i am met with the error that the object is not registered or that the registration has expired.
above this is the message: RG: 167772160 RV:0

System
——————
Intel Nuc Skull Canyon
512GB
16GB Ram
Windows 10 Professional

Kinect2 for Windows
Kinect for Windows SDK 2.0
Visual Studio 2015 etc
Max 7 (registered)

the dp.kinect2-v1.1.20160922.zip was downloaded and installed in the packages folder.

please advise
cheers!

diablodale, 2017-01-04:
Please provide a complete copy of all information in the Max console window. There is information there that can likely help troubleshoot the issue.

Start max
open the dp.kinect2 help file
go to the register tab
Enter your registration name
press register

If it doesn't register, then switch to the Max Window.
Then select all lines of text
copy them, then paste them into this issue please.

jjolivette, 2017-01-04:
there’s nothing else in the console- just the license agreement

jjolivette, 2017-01-04:
hello sir,

below are the contents of the Max window:

dp.kinect2 v1.1.20160922.0 Win 64, Copywright(c) 2014-2016 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those documented in the dp.kinect2 terms at http://hidale.com and granted by purchase/agreement to any entity listed as registered below
-Compiled with Eigen 3.2.8 licensed under Mozilla Public License, v2.0. Source code & license at http://eigen.tuxfamily.org/
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THE SOFTWARE.

dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
viddll.engine version: 1.0.6
dp.kinect2 RG:167772160 RV:0
dp.kinect2 dp.kinect2 is not registered or has expired. Register your key or visit http://hidale,com for purchase and support.

thanks again!

diablodale, 2017-01-04:
The text you pasted above is not the text from the Max Console window. It has been altered.
Please provide the complete unaltered text from the Max Console window.

Meanwhile, the licensing server is not receiving a request from your computer. I suspect there is a networking problem with your computer or your network. Please ensure that you have good networking from your computer to the Internet.

jjolivette, 2017-01-04:
howdy,

the internet connection seems fine
i’ve tried 1 wired connection and 3 wireless connections

here is the text copied directly from the Max Console:

dp.kinect2 v1.1.20160922.0 Win64, Copyright (c) 2014-2016 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those documented in the dp.kinect2 terms at http://hidale.com <http://hidale.com/> and granted by purchase/agreement to any entity listed as registered below.
-Compiled with Eigen 3.2.8 licensed under Mozilla Public License, v2.0. Source code & license at http://eigen.tuxfamily.org/ <http://eigen.tuxfamily.org/>
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com <http://hidale.com/> for purchase and support.
viddll.engine version: 1.0.6
dp.kinect2: RG: 167772160 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com <http://hidale.com/> for purchase and support.

diablodale, 2017-01-04:
Is your computer clock accurate? I am seeing a few attempts to valid licenses from a computer that does not have an accurate clock.

diablodale, 2017-01-04:
Is your computer clock accurate? I am seeing a few attempts to validate a license from a computer that does not have an accurate clock

closed 2017-01-04 by jjolivette

diablodale, 2017-01-04:
Glad you are able to register.
I have some work to do. The server sent a useful message about your clock, but a code issue is preventing you seeing it. I understand and can reproduce this now. I'll make a code change to enable the message to be correctly seen.
Thank you for your patience to isolate the issue.

jjolivette, 2017-01-17:
Hey Dale

I’m getting the same not registered error again despite having corrected the clock
Please advise

cheers,
jean-pierre

diablodale, 2017-01-17:
Please open a new issue unless you believe it is clock related. In the new issue do all the same again as I need a new setup, scenario, config, etc. Remember to read and follow the instructions on how to open an issue.

If you believe it is clock related, then keep it here and do the same. I need all new information.

diablodale, 2017-01-17:
Also, try the newest version of dp.kinect2. I did the work I said I would do. I added more error messages so perhaps you can resolve your own issue with the new messages.
Download at http://hidale.com/shop/dp-kinect2/