diablodale/dp.kinect #40: dp.kinect2: data unavailable for request

noxmox, 2017-02-26:
I can't get the Kinect to work on my new windows installation. The dp.kinect2 external was working great before formatting the HD. Like always I followed the installation guide from Hidale.

But now the Kinect starts and I can see the RGB and the depth image for like 3-4 seconds and then it stops with these messages:

 "dp.kinect2: data unavailable for request"
"dp.kinect2: Kinect has not been successfully opened and initialized"

win pro 64 bits, (updated)
Max 7.3.2 (registered)
KinectRuntime-v2.0_1409-Setup

Any help will be much appreciated.

diablodale, 2017-02-26:
This type of behavior suggests a driver or hardware problem.

"data unavailable for request" is a friendly way of saying "the driver is not providing any data for this frame and there is no possible retry or recovery for this frame".

If you then get an error suggesting the Kinect connection was automatically closed, like you pasted above "...not open", then the errors from the driver were so severe and unexpected that no continued use is possible.

These kinds of behaviors are extreme. I recommend you troubleshoot core problems with the OS, computer, drivers, and sensor. You can also use as part of your troubleshooting the Microsoft Kinect Verifier tool. Step 1 of my setup guide will lead you to Microsoft and that tool. https://github.com/diablodale/dp.kinect2/wiki#setup-guide

noxmox, 2017-02-27:
Dale, thanks a lot for your ultra fast and outstanding support! 
I found an update for my USB drivers and I got it working now... :)
Grüße!

closed 2017-02-27 by noxmox