diablodale/dp.kinect #41: Error 126 loading external dp.kinect

mattgilbertnet, 2017-03-01:
When opening the dp.kinect patcher, I get this message in the Max window: 
"Error 126 loading external dp.kinect"

If I ignore that and attempt to register using the trial key:
I click the bang by "Click to Start", it shows X.
I click "register", nothing happens. No file open window or anything to select the registration key, and no further error.

I'm on a Windows Surface laptop (so, 64bit) running Windows 10 Pro, Max 7 running in Demo Mode, dp.kinect-v1.0.6.

I've added the dp.kinect folder in "File Preferences" and restarted, same thing happens. The only similar issues I've found were from 2013, and were fixed with an update that supported 64bit, or something along those lines.

diablodale, 2017-03-01:
That error is coming from Max, which I think gets it from Windows. The error is something related to not being able to load needed libraries and drivers. This is very likely a setup and/or Windows driver problem.

Please follow the setup guide here https://github.com/diablodale/dp.kinect/wiki
From what you write above, you haven't followed steps 3 and 6.
Please also ensure that Windows Update has no uninstalled drivers or windows components.

Additionally, Max may have unknown-to-me restrictions in their demo mode. You may need to go to Cycling74 for support if they restrict what you can do in their demo.

mattgilbertnet, 2017-03-02:
Sorry, I should have mentioned, I went through those setup steps. I do have newer version of the Kinect Runtime, though, Kinect SDK 2.0-1409. 

I put the files from the zip in a common directory (~user/Documents/Max 7/Library/dp.kinect-v1.0.6), and made sure this was included in Max's File Preferences.

I'll reach out to Cycling74 as well.

mattgilbertnet, 2017-03-02:
Got a quick reply from Cycling74 basically saying they can't help much, but "Max demo shouldn't be interfering with loading the objects". 

They also said to check both Max and the external were 64 bit. I'm running Max 7 64 bit, and I gather dp.kinect supports both (from the filetypes), right?

diablodale, 2017-03-02:
Please visit again the setup at https://github.com/diablodale/dp.kinect/wiki
It appears you have not done each step. For example, in step 3 it guides you to install one of the 1.x Kinect runtimes. Yet, you write above you installed 2.0v1409. That won't work because that's a runtime for the Kinect v2 sensor (in which you would use dp.kinect2).

There is a DLL or library somewhere that is needed (e.g. the runtime I write about above). I can't know because Max isn't providing a more useful message. Windows has some tools in it that might help. Like the windows 10 reliability report. Or the Sysinternals Process Monitor.

dp.kinect.mxe64 supports Max 64-bit.
dp.kinect.mxe supports Max 32-bit.
Both are in the download as well as dependent files. You write about setting preferences and moving files. I recommend keeping it simple (Step 6). Only after that, create a custom install which is for your needs.

mattgilbertnet, 2017-03-02:
Very sorry, somehow I didn't notice dp.kinect2. Just installed it and it's working great! I'll go ahead and close this issue. Thanks for your help!

closed 2017-03-02 by mattgilbertnet

diablodale, 2017-03-02:
Good to hear it's working for you.  Happy patching!