diablodale/dp.kinect #42: Record depth + video to replay in dp.kinect2

aartcore, 2017-03-09:
Hey,

Is it possible to record the depth and video data to load and replay in the object?
In the jit.openni object you can load your recorded .oni files which is very handy when your are testing and programming. Is there also something similar for the dp.kinect2 object?

diablodale, 2017-03-09:
Hello. Microsoft provides tools for recording and playback of the raw data for the Kinect v1 and Kinect v2 sensors. The tools are both called "Kinect Studio". These are developer oriented tools which you can download from Microsoft in the Kinect for Windows Developer Toolkits.

Microsoft sometimes changes their download links. Here are the links that work today.
* Kinect for Windows v1 Developer Toolkit https://www.microsoft.com/en-us/download/details.aspx?id=40276
* Kinect for Windows v2 SDK https://www.microsoft.com/en-us/download/details.aspx?id=44561

aartcore, 2017-03-09:
Ah thanks for the links! Is it then also possible to play the recorded files with the dp.kinect2 object in max?

diablodale, 2017-03-11:
Yes. Both of those tools allow for playback and record. They work independently for all applications by operating directly on the drivers...simulating a sensor at the very low level. Therefore, you will be able to use them with Max.

closed 2017-03-15 by diablodale

aartcore, 2017-04-05:
The Kinect Studio option works to replay the skeleton data, but i didn't manage to get playback the video in the dp.kinect2 object. I test with recording all the different streams but with no luck see the video (rgb or infrared) in the dp.kinect2 object. Is there set with streams which match with the dp.kinect2 object to see if i playback my recording?

Thanks in advance

diablodale, 2017-04-05:
Kinect Studio is a complex developer tool and outside my realm to provide support. If Kinect Studio is setup correctly and your desired streams are recorded and also played back, then all Kinectv2 software (including dp.kinect2) will work. I have tested this repeatedly.

Microsoft can provide you support for Kinect Studio. I recommend you contact them to get the answers on how to use it. https://developer.microsoft.com/windows/kinect

Or try the Kinect v2 forums at https://social.msdn.microsoft.com/Forums/en-US/home?forum=kinectv2sdk

aartcore, 2017-04-05:
Ok thanks,

I will check it with microsoft.

https://developer.microsoft.com/windows/kinect

com/Forums/en-US/home?forum=kinectv2sdk