diablodale/dp.kinect #43: cannot unregister license

hollyhook, 2017-03-10:
I'm on a Win 10 x64 PC with registered Max 7.3.2. I tried to unregister with the help of the demo max patch, in order to move dp.kinect to a new PC. 

See attached screenshot. I selected unregister 2x, got error message dp.kinect ist not registered ...
next I tried to register, which worked.
Then again unregister, which failed with same error message.

[image]

Please unregister my license, so i can use it with my new PC.

diablodale, 2017-03-10:
You are already successful unregistering when you see the message "dp.kinect is not registered...". Nothing more to do.

I can see now that this message can be unclear. I'm describing the state (dp.kinect is not registered) rather than giving direct feedback on the action (e.g. dp.kinect has been unregistered).

I'll make a change in the next version of dp.kinect to make it clearer. :-)
Thank you for choosing dp.kinect for your solutions.

hollyhook, 2017-03-11:
Okay, now I got it. Thanks!