diablodale/dp.kinect #44: The device attached is not a valid Kinect.

nesifao, 2017-04-03:
Hi Dale,

I have a problem with my kinect v1. When I launch your object it works around the first 3 minutes and then I have this message : The device attached is not a valid Kinect.

Then I have to relaunch max to make it works and same message after 3mn. What should I do?

I use your v1.0.6 version of dp.kinect in max 7.3.3 and I have the sdk 1.8 installed on windows 10. 

Thanks in advance.

diablodale, 2017-04-03:
That is an error directly from the Microsoft Kinect driver. Technically it is the error:  E_NUI_NOTGENUINE

I did some searching on the Internet. I see that several people encountered this error when there is something not working well with the USB port or USB controller. All of them fixed that error when they fixed their USB problems.

Here are examples:
http://traumabot.blogspot.de/2013/02/kinect-colour-ir-depth-image-reading.html
https://social.msdn.microsoft.com/Forums/en-US/d11b7c3b-3cf4-4d62-af0e-2ea335a74695/solved-kinect-sdk-10-or-15-feed-in-the-sample-apps-freezes-after-10-to-20-seconds?forum=kinectsdk

I recommend you focus on USB controller and port problems. Remove all other devices on that USB controller. Please remember that a single USB controller usually has multiple ports including unreachable internal ports like keyboards, touchpads, bluetooth, etc. You can see which devices are connected to the same USB controller using the Windows Device Manager and use the menu "View devices by connection".

nesifao, 2017-04-05:
Hi Dale,

Thanks for your quick answer. I tried to update all my usb drivers, disconnect everything in the usb ports, change all the power settings and never turn off or stand by usb ports and I still have the same error happening. The weird thing is few month ago it worked perfectly....

diablodale, 2017-04-05:
I recommend you contact your computer manufacturer or Microsoft. This is a hardware or driver problem and outside the dp.kinect software. This error is direct from Microsoft's drivers and it is impossible for my software to cause the error.

You could have a failing USB cable, failing controller, failing Kinect, or bad power supply on your computer or the Kinect. These are all hardware problems.  Or you could have a new software, drivers, or operating system change that was not there in the past. Because this is an error direct from the Microsoft driver, the problem must be driver, OS, or hardware related.

You can use typical hardware troubleshooting techniques. I always start by trying everything on a second PC and not using any of the same hardware. If it works on the 2nd PC, then it is problem a problem unique to the 1st PC. And then continue to isolate on the 1st PC...

nesifao, 2017-04-05:
I tried on a new computer and it works. I'm pretty sure you are right and it comes from a hadware / driver problem. If I found why I'll let you know. 

Thanks for your help.

diablodale, 2017-04-05:
Good to read. Happy patching!

closed 2017-04-05 by diablodale