diablodale/dp.kinect #46: dp.kinect not registered after writing registration name and opening .dpreg

ondeorange, 2017-07-26:
Hi Dale,
dp.kinect is still not registered after writing registration name "Edgar Hemery" in the register tab and opening the .dpreg i've purchased. I believe I've correctly followed every steps (including 4, 5 and 6). I've also tried opening the trial version with the trial dpreg and I had no results either.
Please help !

diablodale, 2017-07-26:
Hello. Please read the instructions at https://github.com/diablodale/dp.kinect/issues/18 on how to open an issue. I want to help you. To do that I need this information as I previously requested.

Please note, if you are seeking support for dp.kinect, then you are in the right place.

If you are seeking support for dp.kinect2  <--- that's a 2 as in the 2nd Kinect...as in the software you purchased, then the place to search past resolved issues and open a new issue is https://github.com/diablodale/dp.kinect2/issues

diablodale, 2017-07-26:
closing and moving to dp.kinect2 area

closed 2017-07-26 by diablodale