diablodale/dp.kinect #48: unable to  register

MaorEfrati, 2019-02-26:
## Description
Hello, I just brought the dpkinect, and I can not register
max gives the massage:
"
dp.kinect: dp.kinect is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.
"
before I brought it it worked with the trail version, Now both are not working.

## Setup
I'm using an Acer Predetor 17 laptop with Windows 10.
16 ram, 7th gen I7 possessor, with G-force GTX 1070

KinectSDK-v1.8-Setup

Max 8.03

dp.kinect-v1.0.6 

## Steps to reproduce

first time I downloaded it and put it in my Max8/packages folder .

1. StartedMax 8
2. Open the dp.kinect help patch
3. kept the trail name.
4. Press "register"
5. select the trial.dpreg file.
6. It worked just fine.
7. I downloaded the fface tracking dll, and aded it to the folder.
8. payed for the object.

then...
1. Start Max 8
2. Open the dp.kinect help patch
3. add my reg name to "Type registration name below"
4. Press "register"
5. select the file I downloaded from the email you sent
6. the massage ""
dp.kinect: dp.kinect is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.'
shows up in the Max console window,
7. it shows up again everytime I Use "<-- Click to start'

## Result

the "dp.kinect @depthmapres 1 @depthmap 2 @type float32 @colormap 1 @skeletonformat 1 @playermap 1 @skeleton 1" object dose not work.

```
dp.kinect: dp.kinect is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.'
```

diablodale, 2019-02-26:
Hello. All the services are working fine. This is almost always a local issue that is easily resolved.
Please be sure that you follow steps 2 and 3 at https://github.com/diablodale/dp.kinect/wiki#register-dpkinect

You must type the registration name exactly as it is presented in the email you received. Every character and space must be exact.
Also be sure you are not selecting the ZIP file. You want the registration file(s) that are inside the ZIP file.

If you continue to have issues, please provide all of the messages in your Max console. I need _all_ of them to troubleshoot further.

MaorEfrati, 2019-02-26:
I did type the registration name exactly as it is presented (copy-paste from the Email)
maybe the name I chose was too long?

and I am not selecting the ZIP file, I select "dp.kinect-1155.dpreg'

this is what I get when I open it:

ejies  3.2.4  (12/2015)   
     by Emmanuel Jourdan, e--j dev 
dp.kinect v1.0.6.0, Copyright (c) 2011-2014 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those: a) documented in the dp.kinect terms at http://hidale.com,
-and b) granted by purchase or agreement to any entity listed as registered below.
-Includes some derivative work based on (c) Microsoft source code licensed under Apache License Version 2.0.
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.
dp.kinect: dp.kinect face tracking requires FaceTrackData.dll version 1.7.0.510 for your platform (x86 or amd64)
dp.kinect: dp.kinect is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.

diablodale, 2019-02-26:
I suspect the problem is with your registration name. I recommend you copy and paste it from the email. And be absolutely sure you have no extra spaces at the beginning or the end. And that you have no hidden spaces in that text field in which you type your registration name.
I tried here on two different test machines with your registration name and dpreg file. It worked immediately.

diablodale, 2019-02-26:
If you continue to have problems, perhaps a remote control session so that I can register it myself. Or see the problem myself. We can probably find some technology to do a remote control.

MaorEfrati, 2019-02-26:
alright,
Can you e-mail me the exact way you wrote it? if it won't work we can try the remote control thing

diablodale, 2019-02-26:
What is "it"? I don't understand what you are requesting.
If you are asking for your registration name, it was sent to you in email. And in that email it is on a single line. You will need everything on that single line with no extra spaces at the beginning or the end.

diablodale, 2019-02-26:
What is "it"? I don't understand what you are requesting.
If you are asking for your registration name, it was sent to you in email. And in that email it is on a single line. You will need everything on that single line with no extra spaces at the beginning or the end. I just counted and it is 29 characters

MaorEfrati, 2019-02-28:
I'm not sure what I did differently, but it worked!
thanks :)

diablodale, 2019-02-28:
Good news. I will close this issue as resolved.
happy patching!

closed 2019-02-28 by diablodale