diablodale/dp.kinect #49: No Kinect chosen for output.

RaitoWolf, 2019-03-29:
## Description

No Kinect chosen for output.

## Setup

* Dell Inspiron 15 7559, 16 GB RAM, 4 GB GTX 960m, Intel i7 6700HQ
* Windows 10
* KinectSDK 1.8
* Max 7.3.5 registered
* 1.0.6 trial
* Xbox v1 1414 Kinect

[image]

## Steps to reproduce

1. Start Max 7.
2. Open the dp.kinect help patch
3. Click on "Click to start"
4. Error shows up.
5. Cannot change hardware to 1: xbox v1

What result did you *actually* get? Please provide screenshots when helpful.

## Expected Result

I expected it to work, nothing special to write here.

## Workarounds

Nope. I checked the other threads but they didn't help.

## Max console output

[image]

diablodale, 2019-03-30:
This is a duplicate of many previous issues. Please read these https://github.com/diablodale/dp.kinect/issues?utf8=%E2%9C%93&q=is%3Aissue+chosen+for+output

diablodale, 2019-04-01:
Closing this as a duplicate, known issue, caused by Microsoft's licensing restrictions.

closed 2019-04-01 by diablodale