diablodale/dp.kinect #50: Registration issue

mattiatrabucchi, 2019-05-23:
## Description

Please provide a clear and concise description of your issue.

Max console print this error when i try to activate using my .dpreg

dp.kinect: dp.kinect is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.
and after use @verbose 1
dp.kinect = RG: 33558528 RV:0

is the FIRST TIME i buy dp.kinect and try to install on this computer.

## Setup

WIN 10
i76900K
gtx 2080i

* Description of your computer hardware

* Version of your Operating System and any patches/updates

* Version of Microsoft's Kinect drivers (or SDK) you have installed

the last one sdk 1.8

* Version of Cycling74 Max you have installed and if it is registered

8.0.5

* Version of the dp.kinect external you are using

dp.kinect 1.0.6
* ...

## Steps to reproduce

Please detail every step needed to reproduce your issue.
Set `@verbose 1` on dp.kinect to receive additional debug information in the Max console.
I recommend you create a small patch that demonstrates the problem. Attach your patch to this issue.
Here is an example of detailed steps...

1. Start Max 8.
2. Open the dp.kinect help patch
3. insert my registration name 
4. press register
5. choose the .dpreg received by email6
6. get this error

dp.kinect: dp.kinect is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.
and after use @verbose 1
dp.kinect = RG: 33558528 RV:0

## Result

What result did you *actually* get? Please provide screenshots when helpful.

## Expected Result

What result did you *expect* to get?

## Workarounds

Did you discover any workarounds? What are they?

## Max console output

```
When you set `@verbose 1`, additional debug messages were sent to the Max console.
Please copy and paste here...
1. The first group of messages from dp.kinect which provide version information and settings.
2. Include any diagnostic messages you find related to your issue. They will be in yellow/red color.
```

diablodale, 2019-05-23:
This looks like a duplicate of https://github.com/diablodale/dp.kinect/issues/29
Please refer to that issue.
Please also remember, in step 3 of https://github.com/diablodale/dp.kinect/wiki#register-dpkinect, you can not select the ZIP file. You must first extract the DPREG file from the ZIP file.

mattiatrabucchi, 2019-05-23:
Hallo,

i’ve already read that issue.

I’m not trying to read de dpreg from .zip, all the pack is  unzipped and inside default max packages folders.

my registration nama from email is Mattia trabucchi 009090550146

my .dpreg file is named dp.kinect-1203.dpreg

diablodale, 2019-05-23:
Very unusual. I just now used your download and your registration name and it worked on two different computers that I have here.

I believe this is something isolated to your setup.
Please provide clear and very detailed steps that you follow. Examples are above.
Please also provide all the contents of your Max console window.

mattiatrabucchi, 2019-05-23:
mmm. 

i tried asking  the code from my friend.  and that works. so the problem seem solved.

if is ok for you. i keep using this. 

in this way I regularly pay the license for this machine even if i dont use that code.

diablodale, 2019-05-23:
Licenses are limited to one computer. If you use it on two or more computers, the licenses may be revoked and it stop working on all the computers.

The registration process for dp.kinect is very reliable. I continue to believe this is a problem isolated to your specific setup. And we can solve that. I still need the two pieces of information I requested.

mattiatrabucchi, 2019-05-27:
I understand but now that my installation is running, I would not be very inclined to risk playing with licenses with the risk of blocking everything.

---------------------------
www.mattafunk.com
Soundesign & Dj
+39 338 9395297

closed 2019-06-03 by diablodale