diablodale/dp.kinect #52: No skeleton joints output, Max console: 'data unavailable for request'

bartatgithub, 2019-12-17:
## Description

All works except the skeleton joints, Max error: 'data unavailable for request'. 
Search the internet but haven't found a solution. 
Followed all steps carefully from the Wiki at https://github.com/diablodale/dp.kinect/wiki. 
Now dp.kinect is in trial mode, when it works I will buy the license. 

## Setup

UDOO Bolt V8 (special bought for this project) 
- AMD Ryzen™ Embedded V1605B Quad Core/eight Thread @ 2.0ghz (3.6ghz Boost)
- 16Gb 64-bit So-dimm Sockets With Ecc 2400 Mt/s
- Amd Radeon™ Vega 8 Graphics (8 Gpu Cu)
Windows 10 x64 with all current updates
Installed Kinect 1.8 SDK
Installed Max 6.1.10 x64 and it is registered
external: dp.kinect-v1.0.6 trial mode

## Steps to reproduce

connect Kinect v1 XBOX 360 1414
boot Windows 10
start  Max 6.1.10
open example patch from dp.kinect-v1.0.6 trial mode
put all on

## Result

No skeleton joints output, Max console: 'data unavailable for request' 

## Expected Result

Skeleton joints output, no error. 

## Workarounds

Not yet. 
Doesn't work for me because Kinect stays connected: 
https://github.com/diablodale/dp.kinect/issues/40

## Max console output `@verbose 1` set

```
Max console: 'data unavailable for request' 
```

diablodale, 2019-12-17:
Hello. Kinect Gen1 model 1414 was never fully supported by Microsoft on the Windows platform. I discuss this in https://github.com/diablodale/dp.kinect/issues/33 and unfortunately Microsoft's material on the Gen1 sensor has gone dark.

Still, there are some scenarios that might work with 1414. To help you, I need to know exactly what you are doing when things go correctly and when they fail. It would be helpful for you to give me a patch that demonstrates the failure. Then give me click-by-click instructions.

Turn on `@verbose 1` on dp.kinect and try to reproduce your problem again. Share with me your Max Console output. I need to see everything from the copyright with dp.kinect, registration, errors, etc.

With more information I can try to reproduce it on my computer and sensor. I think my 1414 is in the cellar somewhere. I'll look this afternoon.

From where did you install "Kinect 1.8 SDK"? Please share with me the exact url of the download page.

As I wrote, Microsoft has gone dark with Kinect gen1 sensor, so some of the requirements are harder to find....

* Ensure that you have NOTHING else on the same USB2 controller. The Kinect needs 100% of the USB controller's bandwidth. In windows, you can check this with Device Manager, view/devices by connection, and then find your Kinect and verify that the controller to which it is connected has no other devices on it.
* Turn _off_ all your power saving feature. Powermode=high performance.
* Don't use a USB hub. Connect the Kinect directly to the computer.

bartatgithub, 2019-12-17:
Hi and wow, your a fast one. Thanks, here we go: 

### Seams to work now, but error message is still in the Max console?  
The OSC/route skeleton outlet sends info out, see screenshot: 

### screenshot
[image]

### To get rid of the ongoing errors, i send you still all info: 

### _… give me a patch that demonstrates the failure._
For now, and being sure that's not an error of mine, i only use your example patch dpkinect.maxpat that's in dp.kinect-v1.0.6. 

### _Then give me click-by-click instructions._
connect Kinect v1 XBOX 360 1414 to USB 3.0 port
boot Windows 10
start Max 6.1.10
open example dpkinect.maxpat from dp.kinect-v1.0.6 trial mode
click the X <-- Click to start button
click the @verbose 1 to activate the verbose

### _Max Console output_ 
Jitter initialized
dp.kinect v1.0.6.0, Copyright (c) 2011-2014 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those: a) documented in the dp.kinect terms at http://hidale.com,
-and b) granted by purchase or agreement to any entity listed as registered below.
-Includes some derivative work based on (c) Microsoft source code licensed under Apache License Version 2.0.
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect trial has approximately 14 day(s) remaining
dp.kinect: dp.kinect face tracking requires FaceTrackData.dll version 1.7.0.510 for your platform (x86 or amd64)
dumpout1: open 1
dp.kinect: data unavailable for request
dp.kinect: data unavailable for request
dp.kinect: data unavailable for request
dp.kinect: data unavailable for request
...

### _From where did you install "Kinect 1.8 SDK"?_
https://www.microsoft.com/en-us/download/details.aspx?id=40278

### _Ensure that you have NOTHING else on the same USB2 controller._
Generieke USB hub
- Kinect for Windows Camera
- Kinect for Windows Device

### _Turn off all your power saving feature. Powermode=high performance._
Done. 

### _Don't use a USB hub. Connect the Kinect directly to the computer._
Never use that.

bartatgithub, 2019-12-17:
… I think that it works: I get for example a dump out like this: 
`dumpout1: skel 2 l_hand -0.19125 -0.326796 1.022518 1.`
So it seams ok, no? 
Pity that i don't know what did the trick. Tried with power saving on and off gives the same result. 

The only weird thing is that the dump shows only user 2 and the errors are still going on, except of coarse when i put verbose 0.

diablodale, 2019-12-17:
Good news :-)

https://github.com/diablodale/dp.kinect/issues/40#issuecomment-282565759
When @verbose is on, you may occasionally get a `data unavailable for request`. If you get 30 errors/second that is a problem. But if you get 1-2 errors in 3-5 seconds, then this is just the occasional gaps in availability of data from/to the Kinect sensor due to hardware timings, judder, CPU bandwidth, etc. 

You will need to learn from the https://github.com/diablodale/dp.kinect/wiki, help files, and examples for dp.kinect to understand the `skel` message, how the Kinect sensor calculates `user` IDs, and more. 🙂 Both of the examples (skel and user) you gave are working as intended.

dp.kinect (for the Gen1 Kinect) and dp.kinect2 (for the Gen2 Kinect) function very similar. And it is usually easy to use patches on them both. Perhaps look at the tutorials that I made for dp.kinect2 https://github.com/diablodale/dp.kinect2/wiki/Tutorials. They will work on dp.kinect with almost no changes. You can also download dp.kinect2 from my store and look at the help file in that download. You can often change `dp.kinect2` to `dp.kinect` in those patches and they will work on your Gen1 setup.

diablodale, 2019-12-19:
Closing this issue. Occasional `data unavailable for request` warnings are typical behavior with the Kinect and its drivers.

closed 2019-12-19 by diablodale