diablodale/dp.kinect #55: kinect v1 doesnt get recognised

BRUTUS-et-tu, 2022-06-06:
Hey! 
first of all, thanks for all the good documentation and already existing help for this, older, file.

Unfortunatley, after trying all suggested actions; turning off memory integrity, using sdk 1.8 instead of 2.0, my kinect v1 still doesnt get recognised from my PC.
In my device Manager, there is a section called ''other devices'',
 where ''kinect for windows security'' is featured, but nothing else, (camera, audio)

I've tried alot and last year when i first set it up with my computer everything went fine, could it be because the support for windows for kinect has been discontinued? 

I would be happy over a reply, 
thank you!

BRUTUS-et-tu, 2022-06-07:
Description
My windows 10 machine, doesnt detect the kinect v1 scanner.
I followed all workarounds provided; turned off memory integrity, installed sdk 1.8 instead of 2.0,
moved the kinect specs in the device manager. 

this is essentially the problem, i think if I manage to get my kinect seen, i wont have problems with the max patch itself..

Setup

    gtx 1080 graphic caard
    Aorus x570 elite
    amd ryzen 7 3700 8 cores
    32 GB RAM

    windows 10, latest update
    currenttly 1.8 i've tried everyone i could find though 
   8.3
    Version of the dp.kinect external you are using: latest version from github
    ...

Steps to reproduce

Please detail every step needed to reproduce your issue.
the reproduction in this case is not possible for me to show, since the kinect doesnt get recognised by my windows system..

I hope this helps! 
thanks

diablodale, 2022-06-07:
I don't have all the information, but from what I can see, this is a Microsoft/hardware problem and not dp.kinect. If Windows does not detect your Kinect, that is a Windows and Kinect problem -- both which are supported by Microsoft.

Are you using an XBox360 Kinect?  Remember that those are for development only and Microsoft limits them in their drivers. https://github.com/diablodale/dp.kinect/issues/37#issuecomment-264890253

Is your hardware exactly the same as last year?
Is your Kinect power supply working correctly?

Use v1.8 Microsoft drivers. Newer versions of drivers are for 2nd and 3rd generation Kinect. Not yours.

Some links that might be related
https://github.com/diablodale/dp.kinect/wiki#computer-requirements
https://github.com/diablodale/dp.kinect/issues/54
https://social.msdn.microsoft.com/forums/en-US/74e8471f-b88c-43f3-bca8-2d59fae54d8e/kinect-for-windows-pc-incompatibility?forum=kinectsdk

diablodale, 2022-06-07:
I just now tested. The first generation Kinect for Windows works on my Windows 10 machine (intel based) with curreny windows updates and with memory integrity disabled.
I added a picture of the 4 devices you should see in Device Manager to the link above.

BRUTUS-et-tu, 2022-06-07:
after trying all sdk drivers, i never got hold of all 4 devices..

diablodale, 2022-06-07:
Microsoft is responsible for supporting you on these base device issues.
However, I doubt you will get any support from them as they abandoned the gen-1 Kinect. 
If I were to guess, I suspect a Kinect hardware/powersupply failure, or that you are using an XBox360 Kinect.

diablodale, 2022-06-07:
Moving your post here...

> this is all i see, i installed the kinect for windows, 1.0 for now, will try with other instances
[image]

BRUTUS-et-tu, 2022-06-07:
I am using a 360  1414 kinect, maybe thats the problem, when i connect it to my macbook it works with max, but there is now way access its depth sensor on macintosh osx, everything connected to the kinect scanner is at l

closed 2022-06-07 by BRUTUS-et-tu

diablodale, 2022-06-07:
Microsoft does not have Mac drivers. You must be using a hack/open source driver on your macbook.

To use the Kinect XBox360, Microsoft restricts it to development only and the link I provided above has details about that and how to use it as a developer.

BRUTUS-et-tu, 2022-06-08:
I am using freenect for max msp On mac, 
Skanect just worked without additional drivers..

the links provided are outdated unfortunately…