diablodale/dp.kinect #56: Licensing issue

depalmami, 2022-06-08:
## Description
Hi Dale, i can't transfer the license to an other PC o re-register dp.kinect2 on my PC(on wich previously the plugin was working). I was able to unregister but now i can't reregister or transfer the license.

## Setup

Nome dispositivo	DESKTOP-OKA7B8S
Processore	11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz   2.42 GHz
RAM installata	8,00 GB (7,73 GB utilizzabile)
ID dispositivo	95EBD06E-B1A9-46B0-BF23-C469FEA1276D
ID prodotto	00330-53796-75196-AAOEM
Tipo sistema	Sistema operativo a 64 bit, processore basato su x64

(sorry for the italian)

## Steps to reproduce

Please detail every step needed to reproduce your issue.
Set `@verbose 1` on dp.kinect to receive additional debug information in the Max console.
I recommend you create a small patch that demonstrates the problem. Attach your patch to this issue.
Here is an example of detailed steps...

1. Start Max 8( Version 8.2.2 (85d93eb32e9) (x64 windows))

2. Open the dp.kinect help patch
3. Click on the register tab
4. Press the register message
5. Select the .dpreg file i find in your email (that you've sent me on March 4 of this year)

## Result
dp.kinect2: RG: 33558528 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit https://hidale.com for purchase and support.

## Expected Result

Activate dp.kinect2

## Max console output
dp.kinect2: RG: 33558528 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit https://hidale.com for purchase and support.

## Addictional infos

diablodale, 2022-06-08:
Please copy and paste all of the messages from dp.kinect2. There are 10+ messages that include its versions, libraries, and several error messages.

diablodale, 2022-06-08:
While you get those console messages...
...I also see that code 33558528 suggests either your dpreg file or your registration name are invalid.
Please verify that the registration name you use is the "name" you received in a separate email from my store.

closed 2024-01-25 by diablodale