diablodale/dp.kinect #57: 2 Kinects for Windows v1, is it possible?

tamarziv, 2025-05-24:
Hi Dale, 
Is it possible to run 2 Kinects V1 using your dp.kinect on Max/MSP?
If so please let me know what the restrictions are, since right now I can only run one.

diablodale, 2025-05-24:
Yes. https://docs.hidale.com/max/attributes/idsensor
For the original dp.kinect, use the `@idkinect` attribute.

diablodale, 2025-05-25:
I don't provide support for debugging customer patches. If you have a bug or feature of dp.kinect that isn't working, then please open a new bug and answer all the questions you skipped when opening this issue.

- Microsoft docs https://docs.hidale.com/hardware/sensors/kinect-v1 writes the Kinect v1 is a USB 2.0 device. Also writes, the Kinect v1 needs the full bandwidth of USB 2.0 (which is only 480 Mbps).
- USB 2, 3, 4+ controllers have only one "USB 2.0 controller" within them. You can see the controllers in device manager. Here shows my laptop has three [image]
- https://docs.hidale.com/max/dp-kinect I document in two places the hardware bandwidth limitations. Look for "bandwidth" on that page or use the search at the top of every page.

If you have only one USB controller, then you can have only one Kinect v1. This is hardware limitation of Kinect v1 and USB 2.0.

tamarziv, 2025-05-25:
This is helpful, Thank you for your help!

closed 2025-05-25 by tamarziv

diablodale, 2025-05-25:
Yes, you can have multiple Kinect v1 and multiple dp.kinect on the same computer. The usb 2.0 controller hardware is the limit.
1 controller = 1 Kinect v1
2 controllers = 2 Kinect v1
etc.

You can have an unlimited number of dp.kinect on the same computer. Each one can be connected to its own independent Kinect v1.

tamarziv, 2025-05-25:
Thanks again!